mXlab/crocodile #3: Bug ligne 55 avec code updater du github

etiennemontenegro, 2020-05-28:
Donc, rouler train.py avec le code updater du github me retourne la même erreur en lien avec le data loader.

Commande exécuté : 
(base) C:\Users\Etienne\Documents\GitHub\crocodile>python train.py
 -r 64 
--output-path C:\Users\Etienne\Desktop\output 
--path-to-dataset C:\Users\Etienne\Desktop\crocodile_commented\frames

print   : 
Loading dataset...
Traceback (most recent call last):
  File "train.py", line 55, in <module>
    dataloader = DataLoader(dataset, batch_size=BATCH_SIZE, shuffle=True, num_workers=4)
  File "C:\Users\Etienne\miniconda3\lib\site-packages\torch\utils\data\dataloader.py", line 213, in __init__
    sampler = RandomSampler(dataset)
  File "C:\Users\Etienne\miniconda3\lib\site-packages\torch\utils\data\sampler.py", line 94, in __init__
    "value, but got num_samples={}".format(self.num_samples))
ValueError: num_samples should be a positive integer value, but got num_samples=0

hugobb, 2020-05-28:
Okay that's weird... Je n'ai pas ce bug. Je vais essayer de comprendre ce qui se passe. Est-ce que juste avant:
File "train.py", line 55, in
dataloader = DataLoader(dataset, batch_size=BATCH_SIZE, shuffle=True, num_workers=4)

tu peux inserer print(len(dataset)) ?

etiennemontenegro, 2020-05-28:
Ça retoune 0.. 

est-ce que je dois pointer directement au dosier qui contients les images et non les dossiers et bouger timestamp.csv dans ce dossier 

example: 
 changer  : C:\Users\Etienne\Desktop\crocodile_commented\frames

pour : C:\Users\Etienne\Desktop\crocodile_commented\frames\128

hugobb, 2020-05-28:
Il y a quoi dans le dossier frames\raw ?

etiennemontenegro, 2020-05-30:
Oui, y'a les 29000 frames

closed 2020-06-09 by hugobb