mXlab/crocodile #4: Erreur à la création de l'environnement conda avec le fichier crocodile.yml

sofian, 2020-06-08:
Sur Ubuntu 20.04, anaconda ne semble pas trouver plusieurs des packages.

```
\$ conda env create -f crocodile.yml 
Collecting package metadata (repodata.json): done
Solving environment: failed

ResolvePackageNotFound: 
  - ffmpeg==4.2.2=he774522_0
  - mkl==2020.1=216
  - mkl_fft==1.0.15=py37h14836fe_0
  - tk==8.6.8=hfa6e2cd_0
  - intel-openmp==2020.1=216
  - libtiff==4.1.0=h56a325e_0
  - icc_rt==2019.0.0=h0cc432a_1
  - python==3.7.6=h60c2a47_2
  - zstd==1.3.7=h508b16e_0
  - vc==14.1=h0510ff6_4
  - vs2015_runtime==14.16.27012=hf0eaf9b_2
  - pillow==7.1.2=py37hcc1f983_0
  - mkl_random==1.1.1=py37h47e9c7a_0
  - openssl==1.1.1g=he774522_0
  - pytorch==1.5.0=py3.7_cuda101_cudnn7_0
  - freetype==2.9.1=ha9979f8_1
  - sqlite==3.31.1=h2a8f88b_1
  - scipy==1.4.1=py37h9439919_0
  - numpy-base==1.18.1=py37hc3f5095_1
  - libpng==1.6.37=h2a8f88b_0
  - zlib==1.2.11=h62dcd97_4
  - jpeg==9b=hb83a4c4_2
  - numpy==1.18.1=py37h93ca92e_0
  - cudatoolkit==10.1.243=h74a9793_0
  - wincertstore==0.2=py37_0
  - mkl-service==2.3.0=py37hb782905_0
  - ninja==1.9.0=py37h74a9793_0
  - xz==5.2.5=h62dcd97_0
```

sofian, 2020-06-08:
Utile (?): https://github.com/ContinuumIO/anaconda-issues/issues/9480#issuecomment-394873967

hugobb, 2020-06-08:
I uploaded a file on slack

sofian, 2020-06-08:
Just tried it and it looks like it is working.

closed 2020-06-09 by hugobb