mXlab/crocodile #8: Faire fonctionner la Real-Time Clock (RTC) sur le Teensy

sofian, 2020-06-15:


sofian, 2020-06-18:
L'objectif ici serait d'avoir une résolution en microsecondes. Voir cette disucssion à cet égard pour le Teensy 3:
https://forum.pjrc.com/threads/38304-millisecond-precision-from-Teensy-3*-RTC

Également:
https://github.com/manitou48/teensy3/blob/master/RTCms.ino

etiennemontenegro, 2020-07-14:
Ici je vois deux options pour la RTC. 

Étant donné qu'on est déja connecté en réseau avec un ordinateur on peut utiliser TimeNTP qui sync l'heure à l'internet. Sinon il faut acheter  un crystal ( https://www.digikey.ca/en/products/detail/citizen-finedevice-co-ltd/CFS-20632768DZFB/283736 ) et une batterie 3v pour que le teensy soit plus standalone

https://www.pjrc.com/teensy/td_libs_Time.html#teensy3

eringee, 2020-07-14:
I already have the correct crystals in my studio over here actually.  They
are inexpensive (about 33 cents each) but you are right, if we are already
connected to a computer there is no reason that we couldn't use timeNTP.

I think the crystal makes more sense in our application - We don't really
need to know if it is 2am or 2pm when the data is collected, we just want
to make sure that ms is accurate. The idea with the battery is just that if
you are doing something like an alarm, if the power goes out you would want
the RTC to continue registering the correct time.

Because of this I think that soldering a crystal to the Teensy should
suffice.  Next time you are over here you can pick some crystals up?
Otherwise if we need to order something else from Digikey it will be more
worth it.  The Teensy itself has holes on the microcontroller already for
directly modifying the board.

E

etiennemontenegro, 2020-07-14:
Perfect, i could pass by thursday to pick it up