mXlab/crocodile #18: Élaborer un format pour l'enregistrement des séances de biodonnées

sofian, 2020-06-18:


sofian, 2020-06-18:
Pour chaque session d'enregistrement à mon avis il faut enregistrer à la fois (1) des "méta données" relatives à chaque session (par exemple, la date et l'heure de la séance, la durée totale, le lieu, le nom de la personne enregistrée, etc.); (2) un "log" de toutes les données enregistrées avec des timestamps.

Ce qui est névralgique est de tester qu'on est bien capable d'enregistrer un certain nombre de signaux (je dirais, disons, 4-5 channels de données, probablement en float 32 bits chaque) en temps réel à 1000 Hz.

sofian, 2020-06-18:
Ça pourrait être intéressant en fait de simplement encoder nos informations dans un format déjà établi. Il existe deux candidats:
1. le format EDF (https://www.edfplus.info/specs/edf.html) qui a l'avantage d'être relativement compact mais peut être possiblement difficile à implémenter (un peu vieillot)
2. le format Open Signals développé par Bitalino qui peut être soit en TXT ou en HDF5 https://bitalino.com/datasheets/OpenSignals_File_Formats.pdf

sofian, 2020-06-18:
Par rapport à OpenSignals voir: https://pypi.org/project/opensignalsreader/

sofian, 2020-06-18:
Pour EDF pas besoin de réinventer la roue, sur Arduino on pourrait utiliser une des librairies suivantes:
https://github.com/pkhorsand/EDFlib
https://github.com/idotta/edfio/

Il y en a peut-être d'autres.

sofian, 2020-06-18:
Un autre: https://github.com/macawm/edflib

etiennemontenegro, 2020-06-25:
Example du header implémenter:

#Name: Etienne Montenegro                         Date: 23/06/20
#Duration: 1:00:00                                   Location: Montreal
#Start time: 13:40:34                             End Time: 13:40:34
#Signals: heart , gsr1 , gsr2 , resp    Sample Rate: 1000 Hz

timestamp,heart,gsr1,gsr2,resp

À la suite de ça les données sont enregistrer comme un CSV.

sofian, 2020-06-25:
C'est bien, l'information est là. L'encodage en CSV a l'avantage d'être lisible mais n'est pas le plus compact. Peut-être que ce n'est pas grave mais à 1000Hz ça peut quand même se remplir assez vite. Tu peux faire le calcul pour comparer avec un encodage de chaque channel sur 16 bits (2 bytes) comme c'est fait en EDF. Par comparaison en CSV chaque channel prend 1 byte (pour la virgule) + autant de bytes qu'il y a de caractères nécessaires pour exprimer le chiffre encoder (eg. si les données vont de 0 à 1024 le max est donc de 4 caractères soit 4 bytes; donc 5 bytes par channel vs 2 bytes par channel pour le EDF -- c'est 2.5x plus). 

C'est peut-être pas un gros problème mais... disons pour une heure d'enregistrement on aura 1000Hz x 60 secondes x 60 minutes x 4 channels = 14,400,000 données. En format CSV le fichier prendra 720,000,000 bytes (~72 M) vs 28,800,000 (~28.8M) en EDF.

sofian, 2020-06-25:
Pour écrire des données binaires au lieu de ASCII avec l'objet SD tu peux (je pense) utiliser la méthode [write(buf,len)](https://www.arduino.cc/reference/en/language/functions/communication/serial/write). Eg.

```c++
int16_t value = sensor.getRaw();
file.write((unsigned char*)&value, sizeof(int16_t)); // <-- ici on convertis la variable "value" en un array de unsigned char (ie. des bytes) donc un array de 2 bytes; sizeof(int16_t) va retourner la valeur "2" donc on va écrire 2 bytes sur la carte
```

etiennemontenegro, 2020-06-25:
> C'est bien, l'information est là. L'encodage en CSV a l'avantage d'être lisible mais n'est pas le plus compact. Peut-être que ce n'est pas grave mais à 1000Hz ça peut quand même se remplir assez [...]

C'est à ces résultats la que je suis arrivé lundi lorsque j'ai calculé la différence de taille des fichiers. Je me suis dit que même si le fichier serait plus gros il ne serait pas trop gros et permettrait quand même plusieurs heures d'enregistrement sur une carte sd de 4gig.

etiennemontenegro, 2020-06-25:
> Pour écrire des données binaires au lieu de ASCII avec l'objet SD tu peux (je pense) utiliser la méthode [...]

C'est possible oui, j'avais de la misère lundi avec la méthode write() par contre. En écrivrant les données ascii directement en utilisant la méthode File.print() on arrive quand même à la vitesse désirer ( 1000hz) et ça me semblait plus flexible que d'utiliser write(). On peut surment adapter la code pour utiliser .write() si c'est ce que l'on souhaite