mXlab/crocodile #19: Tester si on peut enregistrer dans le format EDF avec la carte SD

sofian, 2020-06-20:
Voir issues #11 et #18 pour plus d'information. L'idée serait de faire un "spike" pour voir si on est capable d'enregistrer directement les données en format EDF.

Il y a deux façons possibles de procéder:
1. Coder tout "from scratch" en se basant sur la [spécification de EDF](https://www.edfplus.info/specs/edf.html) Avantage: le format EDF est pas si difficile que ça et on garde le contrôle. Désavantage: c'est peut-être plus difficile qu'on le croit et il y a quand même du travail à faire.
2. Utiliser du code d'une librairie EDF existante. Avantage: le code est déjà écrit et on peut penser que la personne qui l'a fait a bien pensé à son affaire. Désavantage: on pourra probablement pas utiliser la librairie "telle quelle" car ils vont utiliser par exemple des file handlers ou des librairies C++ qui sont pas supportées sur Teensy.

Je ne sais pas quelle est la meilleure manière de procéder. Il faut se lancer à mon avis et voir comment ça avance. Peut-être que la meilleure approche sera un mélange de (1) et (2) ie. coder par soi-même mais en allant chercher l'inspiration et des bouts de code d'une autre librairie.

etiennemontenegro, 2020-06-22:
Je vais écrire notre propre version de l'EDF. L'exporter en fichier CSV avec un header en commentaire m'a l'air plus rapide à programmer. On va pouvoir ignorer le header à la lecture avec un technique comme ça : 

https://wuhrr.wordpress.com/2016/02/16/process-csv-with-comments/

Sinon on peut commencer à lire le fichier à une ligne définie.

eringee, 2020-06-22:
Le seul probleme avec le csv cest que ca prends au moins 2x plus du space
disk.  Peut etre cest pas grave quand meme on pourrais acheter les cartes
sd au 64gb...

Make a calculation about how much space it will take to record an hour of
data with 4 channels..for edf versus csv.

Une alternative serait d'emregistrer en EDF seulement les donnees pas le
header. 16 bits par channel.