mXlab/crocodile #20: Intégrer enregistrement des données + envoi OSC (le tout à 1000 Hz)

sofian, 2020-06-20:
Ce issue devrait être fait après avoir résolu les issues suivants:
1. Intégrer et tester l'enregistrement de données à 1000Hz sur la carte SD #11 
2. Intégrer la communication avec le Teensy via RJ-45 / OSC #16
3. Faire fonctionner la Real-Time Clock (RTC) sur le Teensy #8
4. Tester si on peut enregistrer dans le format EDF avec la carte SD #19

L'idée est de tout mettre ensemble et de valider que ça roule toujours à 1000Hz!

etiennemontenegro, 2020-06-22:
J'ai réussi à implémenter la vitesse désiré , mais je crois qu'il y a place à de l'optimisation. Je souhaiterais vérifié avec toi à ton retour de vacances c'est pourquoi je l'aisse une note ici.

eringee, 2020-06-22:
Hi etienne just make sure to commit your code as you go thanks!

eringee, 2020-06-22:
Sounds great as well! B)