mXlab/crocodile #21: Les enregistrements s'arrêtent aléatoirement.

etiennemontenegro, 2020-07-01:
Les enregistrements s'arrêtent de manière aléatoire. C'est comme si par moment, le teensy perdait sa connection usb à l'ordinateur.

Sur 8 enregistrements test qui visaient une durée de 10 minutes, seulement 1 l'à complété. Les autres on finis à :
2min08
7min49
1min43
4min47
5min06
2min11
7min57

Il ne semble pas avoir de corrélation par rapport au moment ou l'enregistrement arrête.
J'ai eu plus tôt cette semaine aussi des bogues  ou le hardware semblait défectueux. Desfois, il n'était pas reconnu et il suffisait de déconnecter le câble USB et le reconnecter et le hardware était reconnu. 
J'aimerais tester avec un teensy qui aurait des headers femelles pour voir si c'est les connexions sur mon breadboard qui cause le problème.

eringee, 2020-07-02:
does this error occur while you have the serial monitor open on your
computer?

etiennemontenegro, 2020-07-02:
Yes, i'm using it to debug. I'll try without

eringee, 2020-07-02:
if the serial window is open the issue could be that the serial data is
jamming up. 1000Hz is way too fast for a hardware serial connection to
handle reliably.  For a serial connection with microcontroller hardware to
be reliable it can only really handle around 100Hz (or a little less
actually, at once every 16millis) So it would be better to just let the
test run on the SD card and let it do its thing, because I think the I2C is
more stable.  If the same behavior persists even after closing the serial
monitor it might be something to do with interrupts

etiennemontenegro, 2020-07-09:
Changing the Teensy for one with stackable headers and using a wiz850io instead of a wiz820io seems to work. 

I was able to do a two hours recording today

closed 2020-07-09 by etiennemontenegro