mXlab/crocodile #22: Carte Micro SD classe 4 seulement, info non écrite desfois.

etiennemontenegro, 2020-07-14:
Je viens de faire des tests toutes la matinée pour vérifier si mon système de marqueur fonctionnait et j'ai remarqué que sur la carte SD que vous m'avez prêté il manque des informations écrites desfois.
J'ai lancé 5 tests de longueurs différentes qui plaçaient des marqueurs à des intervalles différents avec deux cartes sd différentes pour voir si mon hypothèse se tenait. 
Avec la carte SD que vous m'avez passé dans un test il manquait 2 marqueurs. À l'endroit où le marqueur était censé être écrit, il y avait un écart d'environ 90 samples. Tous les enregistrements sur cette carte avaient aussi un écart entre les samples 128 et +- 230. 
J'ai comparé avec une autre carte sd que j'avais à la maison. La classe n'est pas marquée dessus parcontre. Aucun test ne manquait de l'information, n'y les marqueurs , n'y les samples entre 128 et +- 230. 
Une carte sd de classe 4 à une vitesse d'écriture de 4mb/sec
J'ai aussi trouvé ça sur sdcard.org qui expliquerait potentiellement le problème.

Fragmentation and Speed
By repeating deletion and write of files, data area is gradually fragmented and it influences write speed. Generally, write speed to a fragmented area is slower than sequential write speed due to flash memory characteristics. In an era when memory capacity is not large enough, fragmented write needed to be considered. However, high capacity memory card is available at this time, Speed Class write is defined to perform sequential writes to a completely un-fragmented area (called "Free AU"). It makes Speed Class controls of host easy. On the other hand, even unused memory exists in total, there is a possibility that host cannot perform Speed Class recording. In that case, data arrangement to reduce fragmented area or move data to anther storage to re-format the card will be required.

J'aurais proposé d'acheter une carte mirco SD de classe 10 si vous n'en avez pas chez vous.

eringee, 2020-07-14:
Good catch - I mostly have inexpensive cards with low speeds that I used in
robotic installation, not clear if I have class 10.  I will get your
address in slack and order a few of these directly to your place

https://www.amazon.ca/Gigastone-Adapter-MicroSD-Motorola-Smartphone/dp/B01MTW4U8W/ref=sr_1_8?dchild=1&keywords=sd+card+class+10&qid=1594747348&s=electronics&sr=1-8

etiennemontenegro, 2020-07-14:
I sent it to you, 1 would do. We could take two just in case

closed 2020-08-02 by etiennemontenegro