mXlab/crocodile #33: Tester un entraînement de classification à partir des biodonnées 2021 et des features

sofian, 2023-12-28:


sofian, 2023-12-28:
@hugobb Peux-tu jeter un coup d'oeil et me dire ce que tu en penses?

sofian, 2023-12-29:
J'ai réussi à améliorer grandement la performance en ajoutant des features supplémentaires grâce à la librairie "neurokit".

```
Overall Accuracy: 82.49%
Accuracy for label NEU: 42.34%
Accuracy for label ANG: 94.25%
Accuracy for label ARO: 94.36%
Accuracy for label FEA: 85.25%
Accuracy for label HAP: 83.40%
Accuracy for label SAD: 82.31%
Accuracy for label SUR: 95.50%
```

hugobb, 2023-12-30:
Tu utilises combien de features au final ?
Ca pourrait être interessant d'essayer d'analyser quels features sont importantes pour la prediction.
Parce que plus il y a de features plus j'ai peur que le modele ne generalise pas bien a des nouvelles donnees.

hugobb, 2023-12-30:
Les donnes que tu as utilise c'est les nouvelle donnees de Laurence ? J'imagine qu'il n'y a pas la video associe ?
Est-ce que les donnees ont ete recolte sur plusieurs jours ?

hugobb, 2023-12-30:
J'ai essaye de downloader les fichiers depuis le drive mais je sais pas pourquoi Google les convertis en fichier .xlsx et j'arrive pas a les loader avec le code

sofian, 2023-12-30:
Il y a deux modes: "biosppy" (qui utilise 6 features) et "neurokit" (qui utilise 11 features).

sofian, 2023-12-30:
En effet ce sont les nouvelles données, captées avec le nouveau biocapteur, sans la vidéo associée. De toute façon, nous n'avions aucun jeu de données associée à la vidéo qui contienne la respiration.

Les données ont été captées sur plusieurs jours différents.

sofian, 2023-12-30:
Les données fréquentielles (BPM et RPM) sont fiables de jour en jour, mais les autres données le sont sans doute moins et il est probable que ça ne généralisera pas bien. Il faut que je réfléchisse à des manières de standardiser les valeurs autant que possible.

sofian, 2023-12-30:
J'ai zippé les documents dans un fichier csv.zip que tu peux télécharger (à partir du même répertoire de base).

sofian, 2023-12-30:
J'ai refait un test en réduisant les features à des features "invariants" (ie. pas basés -- ou du moins pas tellement affectés -- par les variations d'amplitude).

Voici l'extrait que j'ai modifié:
```python
                ppg_data, _  = nk.ppg_process(data_df[self.HEART], sampling_rate=sampling_rate)
                # ppg_data = ppg_data[['PPG_Clean', 'PPG_Rate']]
                ppg_data = ppg_data[['PPG_Rate']]

                rsp_data, _  = nk.rsp_process(data_df[self.TEMP],  sampling_rate=sampling_rate)
                rsp_data = rsp_data[['RSP_Rate', 'RSP_RVT']]
                # rsp_data = rsp_data[['RSP_Clean', 'RSP_Amplitude', 'RSP_Rate', 'RSP_RVT', 'RSP_Symmetry_PeakTrough', 'RSP_Symmetry_RiseDecay']]
                
                eda_data, _  = nk.eda_process(data_df[self.GSR1],  sampling_rate=sampling_rate)
                eda_data = eda_data[['EDA_Phasic']]
```

Résultats préliminaires moins impressionants mais pas complètement aléatoires:

```
Overall Accuracy: 46.24%
Accuracy for label NEU: 30.34%
Accuracy for label ANG: 79.31%
Accuracy for label ARO: 51.32%
Accuracy for label FEA: 33.55%
Accuracy for label HAP: 33.29%
Accuracy for label SAD: 35.83%
Accuracy for label SUR: 60.11%
```

hugobb, 2023-12-30:
J'ai joue un peu avec le code, et je me suis rappele de quelquechose d'important: les donnees ne sont pas iid (a cause de la dependance temporel). Je pense que c'est problematique pour evaluer la generalisation, parce que c'est fort probable que si on prend 2 instants consecutifs, les 2 samples seront quasiment identique, mais l'un pourrait etre dans le trainset et l'autre dans le testset. Donc c'est quasiment guarantie que la trainset accuracy soit tres proche de la testset accuracy.

Une solution c'est de decouper le dataset en chunk consecutif d'une certaine taille par example 1min, et d'ensuite selectionner une partie des chunk pour le trainset et l'autre partie pour le testset.

sofian, 2024-01-04:
Petites choses à faire:
- normalisation par séance
- appliquer à nos propres données (voir jeu de données générés par Erin)

sofian, 2024-01-05:
Les résultats préliminaires en séparant le training et test set par séance (et non seulement par blocs) sont assez décevant, le réseau ne semble pas en mesure de généraliser. Je dois vérifier si j'ai bien fait les choses (notamment concernant la normalisation) mais bon, ce ne serait pas non plus très étonnant étant donné le faible nombre de jeux de données.